easyw/kicad-action-tools #7: No attribute 'm_AuxOrigin'

dennevi, 2021-12-02:
When pushing Apply om the Generating Fab Positions dialog I get the following error message:

[image]

I'm using version: (6.0.0-rc1-135-gd60474d3ef), release build
Platform: Windows 10 (build 19042), 64-bit edition, 64 bit, Little endian, wxMSW
Build Date: Nov 24 2021 12:00:44

easyw, 2021-12-06:
@dennevi 
this should be solved with package v1.0.4 (fabrication posistion tool v1.2.8)

dennevi, 2021-12-06:
That's strange... That's the version I have and I still get this error. Just checked and the ____version___ variable in fabrication_positions.py says 1.2.8, and the KiCad PCM which I used to install this plugin says that the version is 1.0.4.

easyw, 2021-12-08:
@dennevi 
could you please check in your plugin folder the source file?
C:\Users\Albin\Documents\KiCad\6.0\3rdparty\plugins\com_github_easyw_kicad-action-tools\FabricationPositions\fabrication_positions.py if @line 82 you have this code:
`        getAO = my_board.GetDesignSettings().GetAuxOrigin()`
1) Could you eventually uninstall and reinstall the plugin?
2) Do you have previous installations of the plugin?

dennevi, 2021-12-08:
Yes, that's indeed the code on line 82.
[image]
After checking the line, I did this:
* Uninstalled the plugin from PCM and verified that all files were removed.
* Reinstalled the plugin from PCM and checked that line 82 is still the same.
* Tried the plugin again. Now it works!

I cannot figure out why it works all of a sudden. I did re-install before writing my last comment regarding the version 2 days ago, and it didn't help then. The only thing I've done since then is restarting the PC. 

I've now tried the plugin on two different project (one of which I know it didn't work on before), and I've tried opening the board both using KiCad and in standalone mode. All seems to work just fine!

Really strange... I'll see if I can get the error to pop up again.

easyw, 2021-12-09:
@dennevi 
I'm closing it ... please feel free to reopen it in case of issues

closed 2021-12-09 by easyw