easyw/kicad-action-tools #19: [Snap to grid] - error with 7.0

Martinnnnnnnnnnn, 2023-02-25:
Snap to grid gives an error with Kicad 7.0:
[image]
PCB action tools V1.0.4
Application: KiCad x64 on x64

Version: (7.0.0), release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.12.1
	HarfBuzz 5.0.1
	FontConfig 2.14.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb 12 2023 01:35:19
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	ngspice: 39
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON

DeepCoreSystem, 2023-02-26:
Hi, your screenshot is about the "Missing 3D Models" plugin and not "Snap to grid"!

For the "Missing 3D Models" it is easy... add the corresponding import instruction.
See attached screenshot.
[image]

But now I need help with the snap to grid function!

Martinnnnnnnnnnn, 2023-02-26:
Sorry, I pressed the wrong icon. Here's snap to grid:
[image]
How can I help you? I can upload the project if this of any use.

DeepCoreSystem, 2023-02-26:
Here are the modifications of 2 files that are needed with KiCad 7 for the snap to grid function:

https://github.com/easyw/kicad-action-tools/pull/16/files

Martinnnnnnnnnnn, 2023-02-27:
Are you going to publish this for the package manager?

flowild, 2023-04-09:
sadly the fixes do not work for kicad 7. i get the same error messages as martin even after importing the 2 new files

Martinnnnnnnnnnn, 2023-04-09:
Installing the package from the 1.0.6 tag https://github.com/easyw/kicad-action-tools/releases/tag/v1.0.6 worked for me.