easyw/kicad-action-tools #24: Move to Layer, Possible to work in Footprint Editor?

thestumbler, 2023-06-09:
Just a question, not really an issue.

As the title says, can/could this plugin be made to work from within the footprint editor? 

-Chris

easyw, 2023-07-05:
this unfortunately is not possible AFAIK
plugins don't work on footprint editor

closed 2023-07-05 by easyw

thestumbler, 2023-07-24:
BTW, I figured out a workaround. I was turning a small PCB into a footprint in order to make a tester board. For example, I wanted the board outline on the edge cuts layer to appear as the silkscreen of the footprint. That was what brought about my original question. 

The workaround -- on the PCB, I temporarily duplicated the board outline. Then, using this plugin, moved the duplicate to the silkscreen layer. Finally, did a cut / paste of this silkscreen from the PCB into the footprint editor. Voila. 

There is a script I found online that converts a PCB into a footprint, but it seemed more complicated than needed for my application. I should note that in my case, the board outline was not just a rectangle, but a complicated shape generated in a CAD program and imported into KiCad using a DXF file.