easyw/kicad-action-tools #25: Version in KiCad PCM is outdated

joelspadin, 2023-09-03:
When installing this from the KiCad plugin and content manager, the latest version available is 1.0.4, so issues like #22 still exist. I'm guessing the newest version needs to be submitted to https://gitlab.com/kicad/addons/metadata?

Patronics, 2024-02-27:
I just did a merge request to update the metadata [here](https://gitlab.com/kicad/addons/metadata/-/merge_requests/308), though since I'm not the original creator of the plugin I'm not certain if they'll merge it or not. We should see one way or the other pretty soon though!

Patronics, 2024-02-27:
As I'd suspected, they want a confirmation from the original maintainer that it should be updated, @easyw could you respond to the [metadata merge request on gitlab](https://gitlab.com/kicad/addons/metadata/-/merge_requests/308) to confirm it can be updated (and/or provide your own metadata update for them to release?)