easyw/kicad-action-tools #28: KiCAD v8 compatibility

thisisgregm, 2024-04-27:
Hi,
Any chance to add the compatibility for KiCAD v8 please?
Many thanks.

AgiosAndreas, 2024-05-03:
Actually, this issue was fixed in CODE but wasn't added to release.

EmielV2002, 2024-06-18:
Hello, I have downloaded the latest version of the plugin via Github and installed it via the PCM manager. I have commented out line 157 in *move_to_layer.py*. When I try to change a graphic to another layer I get the following message *select drawings to be moved to new layer use GAL for selecting lines*. I have selected a graphic so I'm not sure what I'm doing wrong. 
[image]

Application: KiCad PCB Editor x64 on x64

Version: 8.0.1, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 15 2024 01:52:47
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:

easyw, 2024-11-10:
@thisisgregm @AgiosAndreas 
I have made v1.0.7 release compatible with kv8
ATM the package can be “installed from file” using the kicad plugin manager.

easyw, 2024-11-10:
@EmielV2002 
I think your issue is related to the object selected. 
I suspect you are trying to move to an other layer some footprint drawings; this is not possible with the tool. You need to eddit your footprint inside the fp editor.