easyw/kicad-action-tools #29: Snap2Grid using current grid

mengstr, 2025-02-10:
It's a really neat tool, but quite often I use a bit of oddball grids and trace widths like 6.25 mil which is a lot fatter than the min specs for cheap fabs and also allows me to fit two traces between the legs of a DIP.  But of course there's no option for 6.25 in the dropdown...

Could you please add a new option that is just "the current grid" to the dropdown?  That would make the tool so much more usable.

thisisgregm, 2025-09-08:
Option to set default choice would be good too. Currently looks like the default one is 0.1mm, so if you use any other grid you have to change it each time from 0.1mm to whatever your grid is.

malcolm-sparkfun, 2025-12-17:
I came to this repo just to suggest the same thing. I am working with an odd grid not available in the drop down and would love to use the `Snap Selected Footprint(s) to Grid` feature to snap to this grid. Either offering the current grid as an option, or allowing users to add grids to the selectable list, or just changing it to numerical input box rather than drop down would all provide this feature. This would make this much more universal.